Integrated-Project-2-2021-UVic-UCC/sun-tracking #2: Shoping list 1:

XaviMico17, 2021-03-08:
-[] Arduino nano
-[] Photodiodes
-[] Photoresistors
-[] AOP
-[] Servomotor
-[] Resistors
-[] Solar panel
-[] LCD Screen

josepmaria-carbonell, 2021-03-09:
In order to fulfill the objectives the motors **can not be** servomotors, you can not scale the device and use them for a bigger solar panel. **Look for dc motors or stepper motors**.  I also suggest to look for a bigger solar panel and a rechargeable battery to do a self-sustaining energy system (electronics must disconnect when not needed).

XavierPlatas, 2021-03-24:
**### Shopping list update (24/3/2021)**
- [ ]  Arduino nano --> XP
- [ ] Photodiodes --> XM
- [ ] AOP --> XP
- [ ] Stepper motor --> XP (1 missing!!)
- [ ] Resistors --> XP
- [ ] Solar panel --> XM
- [ ] LCD Screen --> XP
- [ ] Protoboard --> XP
- [ ] I2C Adapter --> XP
- [ ] PLA --> Jordi
- [ ] Wire --> XP
- [ ] Real time clock --> XP
- [ ] Converter to store the energy

closed 2021-06-04 by nincodinachs2000