Integrated-Project-2-2021-UVic-UCC/sun-tracking #3: Shopping list 2.

XaviMico17, 2021-03-08:


XaviMico17, 2021-03-08:
This was actually created by Nico.

NicoScipione, 2021-03-13:
The changes in the shopping list are the solar panel and the motors.
We change the solar panel for one more powerful and a little bigger in case we can make a feedback mechanism.
We delete the motors of the shopping list because we will try to use the stepper motors that Jordi already have.
I gona send to you @josepmaria-carbonell and @moises-garin the new shopping list via e-mail and do the new check list in this issue.

NicoScipione, 2021-03-13:
- [ ] Arduino nano
- [ ] Photodiodes
- [ ] Photoresistors
- [ ] AOP
- [ ] Stepper motors (supplied by Jordi)
- [ ] Resistors
- [ ] Solar panel 
- [ ] LCD Screen
- [ ] Protoboard
- [ ] I2C Adapter
- [ ] PLA
- [ ] Wire
- [ ] Real time clock

josepmaria-carbonell, 2021-03-15:
Please look at this solar panel, if it fulfils the specifications:

https://www.amazon.es/enjoysolar%C2%AE-monokristallin-M%C3%B3dulo-Jard%C3%ADn-Camping/dp/B076B651Y8

@Integrated-Project-2-UVic-UCC/sun-tracking

XavierPlatas, 2021-03-16:
This solar panel fulfills all the electric specifications, but not the mechanical ones.

We are commenting if it could be made with this one , if not we would look for some more.😊

josepmaria-carbonell, 2021-03-16:
> This solar panel fulfills all the electric specifications, but not the mechanical ones.
>  https://www.amazon.es/enjoysolar%C2%AE-monokristallin-M%C3%B3dulo-Jard%C3%ADn-Camping/dp/B076B651Y8
> We are commenting if it could be made with this one , if not we would look for some more.blush

Which are the mechanical specifications that it doesn't fulfills? Why you can not adapt to those conditions?
(Take into account that your system has to be scalable and to be applied to bigger solar panels)

These are the other alternatives you propose, I still don't know if they can be bought, we will ask it to Jordi tomorrow:
https://tienda.bricogeek.com/placas-solares/883-panel-solar-9w.html?search_query=panel+solar&results=7
https://www.robotshop.com/es/es/panel-solar-pijuice-12-vatios.html 

@moises-garin    can you take a look?

closed 2021-06-04 by nincodinachs2000