Integrated-Project-2-2021-UVic-UCC/sun-tracking #8: Structure of the code

XavierPlatas, 2021-03-21:
[image]

XavierPlatas, 2021-03-21:
@moises-garin @XaviMico17  It is the first version of the code, can you take a look?

XavierPlatas, 2021-03-24:
[Structure_Code_v1.pdf](https://github.com/Integrated-Project-2-UVic-UCC/sun-tracking/files/6200628/Structure_Code_v1.pdf)
Second Version of the structure of the code. Improvements:
- Mix two loops
- More robust
- Send warning in case of malfunction
- Able to exchange to the loop that best suits

moises-garin, 2021-04-21:
Some comments:

1. One hour is a lot of time. May be wake up every 10 minutes? The important is to set at time that limits the sun movement to a certain number of degrees.
2. The logic would be to move according the sun sensor if the amount light is enough. This means, if the panel is already facing the sun (more or less) and the sky is clear. In this situation, the calculation of the sun position might be useful to detect a malfunction of the system (error too high between real sun position and calculations).
3. If the sun sensor is not giving reliable data, move according to the calculations. 

B.R.

closed 2021-06-04 by nincodinachs2000