Integrated-Project-2-2021-UVic-UCC/sun-tracking #11: Mid-Term presentation

nincodinachs2000, 2021-03-25:
[Mid_term_presentation_Sun_Tracking (1).pdf](https://github.com/Integrated-Project-2-UVic-UCC/sun-tracking/files/6202834/Mid_term_presentation_Sun_Tracking.1.pdf)

ibraamar, 2021-03-26:
Good morning,

First, I don't understand really well what is the electric prototype. Seeing that the task will be in progress until 2 of June, I suppose that is the electronic prototype or the prototype of the project as well. 

About the electrical and mechanical domain, I see a short and efficient explication about how you are progressing in the project, with visual and clear slides.

To conclude, I would like to congratulate you for the project progress and I hope a sun-tracking prototype for the final of the course.

Ibrahim Amar - Power Glove Team.

marcsala27, 2021-03-26:
Hey, I talked with some of your team members about your project and I think that it's really interesting because nowadays, renewable energies are really important. I know that you have had some doubts about the motor needed to move the solar panel. I would like to know about the motor that you have finally chosen and why. Thank you!

SergiMuac, 2021-03-28:
Hi everyone!!

First of all congratulations on your job!!
Your presentation was quite clear and well prepared.
Sincerely, we didn't understand what are you talking about when you say that your prototype will be hybrid. We didn't undestand what is the open and close loop. We would like you to explained again more detailed.

Finally it seems that there are some points that are not clear and maybe you should try to focus on them until you'll get them.
We hope you can deal with all the stoppers, and if we can help just tell us something.

Eye-Sphere Team.

nincodinachs2000, 2021-03-28:
@marcsala27  we finally decided to get a stepper motor, so this motor if we want to get a bigger size sun tracking, choosing a servomotor is more difficult to get, so we decided using a stepper motor that is more easy to scale. The torque we need is more or less 0nm because we design a weight compensation. Thank you for your report.

XavierPlatas, 2021-03-28:
Hi @SergiMuac, we appreciate your comments and we will take it into account.
I'm sorry because it seems to me that we do not explain ourselves well, we want to use a system in between a close loop and an open loop.
An Open loop system is a type of continuous control system in which the output has no influence or effect on the control action of the input signal.
[image]
In our case, the system will work in open loop when the photodiodes are not capable of detecting sunlight and we will update the position of our motors by calculating the apparent position of the sun.

And a closed-loop system is a fully automatic control system in which its control action being dependent on the output in some way.
[image]
In our case, we are going to update the position of our stepper motors in function of the value of the photodiodes.
I hope it is understood a little better, if not, do not hesitate to ask again! :)

Could you specify which points are the ones that are not completely clear?

Guillem-Maza, 2021-03-28:
Once we have attended your presentation, we have to say that your explanation about your project was very well prepared and clear.  
We like a lot the way you present and organize the objectives in the presentation, because on this way we could know clearly the parts you have achieve and the ones you don’t.  

We think that is a very interesting project, and we want to know if you are thinking to make this prototype a real product, to be in the markets and prepared for all types of houses.  
Another question that our group have about your project it’s if the structure is designed thinking in the variety of meteorology depending of the zone, for example, the speed of the wind. 

One part we think that maybe it could be improved is the way you put the electrical schematics because it was so hard to understand something because of the size of the photo. 

Above all, we think that you are working hard and you will get the project correctly.  

Keep working and good luck! 
Covid Device team.

nincodinachs2000, 2021-03-31:
@Guillem-Maza 
Thank you for your interess on our project, It is very greatful to hear that. Answering your about the weather problems, in case of water, our electrical components will be protected on a cas, so they cannot get harmful. The Solar panel is waterproof, so it doesn't get affected. Another weather climate is the wind. Our motor has an initial torque that can support the wind, in case of a windstorm it can get harmful, so we are thinking of a way to solve it. Our principal caution, is the snow, so it snows really hard, it can cover the Panel and because of this we cannot get solar energy, we are thinking ways to solve this problems.

Our principal use of the projet is for feeding up a parking meter, but if our project is scaled, it can be used ina house.
I hope this solved your group doubts, and if something is not well cleared don't doubt to get in touch with us.
Yours faithfully,
Sun-Traking.

josepmaria-carbonell, 2021-04-06:
**Good job guys !**

closed 2021-06-04 by nincodinachs2000