Integrated-Project-2-2021-UVic-UCC/sun-tracking #17: -Battery Charger device-

XaviMico17, 2021-04-14:
Greetings,

There is a subject I want to ask about the battery charger device. We dispose from two lead batteries with a max charge current of 0.39A. I have been looking for an intelligent device able to controll the amoung of power and current the battery recieves from the solar panel, but most of the devices I find are for 10A batteries. Which means we can not use them.
Researching I found that lead batteries can be charged with "less" precaution than litium since lead can not explode or ignite easily, which brought me to think, maybe we can find annother way. 
I asked Jordi yesterday morning, that  I was able to be in the lab and he suggested to use a Power and Current regulation circuit or even to buy the charger and use a current regulation circuit to lower the 10A to 0,3A.

I would like to know your opinion about what would be the best solution, which can also be to find other batteries if you think it is too complicated or inefitient to make one of those circuits.

Thanks in advanced

XaviMico17, 2021-04-14:
[image]

This is a picture of one of our batteries

moises-garin, 2021-04-16:
Hello,

the PV charge controller should be specifically designed for Lead-Acid battery, as they take care of safely charging the battery. However, your battery is quite small, and most charge controllers go from 10A or above, so, they could fry the battery. It is not easy to find solar-panel charge controller for such low current <1A.

So, the first think you should do is calculate the peak charge current the system can deliver to the battery when the solar panel is giving the peak power output. You must consider the efficiency of the DC-DC converter that adapts power from the panel voltage to the battery voltage. If this value is similar or equal to the max. charge current, you must do nothing. If the calculated current can go above 0.39A, then you should consider a current limiter or a small charger for Lead-acid batteries and place it between the solar-panel charge controller and the battery.

Other alternative could be just buying a beefier battery.

best regards.

closed 2021-06-04 by nincodinachs2000